nanoframework/Home #1537: [VSCode Extension] msbuild cannot find target "Build" for command "Build Project"

MexHigh, 2024-08-29:
### Tool

Visual Studio Code extension

### Description

I'm trying to build the sample Blinky Project. However, this happens with all projects (even newly created ones).

When running the "nanoFramework: Build Project" command, it fails with the following error:

```
nuget restore "/home/leon/Code/Samples/samples/Blinky/Blinky.sln" && msbuild "/home/leon/Code/Samples/samples/Blinky/Blinky.sln" -p:platform="Any CPU" -p:NanoFrameworkProjectSystemPath=/home/leon/.vscode/extensions/nanoframework.vscode-n
anoframework-1.0.181/dist/utils/nanoFramework/v1.0/ -verbosity:minimal
Automatische MSBuild-Erkennung: Die MSBuild-Version „15.0“ aus „/usr/lib/mono/msbuild/15.0/bin“ wird verwendet.
Alle in "packages.config" aufgeführten Pakete sind bereits installiert.
Microsoft (R)-Build-Engine, Version 16.10.1 für Mono
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

/home/leon/Code/Samples/samples/Blinky/Blinky/Blinky.nfproj : error MSB4057: Das Ziel "Build" ist im Projekt nicht vorhanden.
```

So basicly, msbuild (mono) cannot find the "Build" target, and it is indeed not there. I don't know how to add this manually.

The Build target is also missing from newly created solutions/projects with the nanoFramework extension.

**Additional info**:
- The command "nanoFramework: Flash device" hangs forever after selecting an image version. Maybe this is related to this, I'm not sure.
- Running the `nanoff` command manually to flash a nanoCLR seems to work
- Seems related to #1508 (not sure because of the chinese error messages)

### How to reproduce

1. Open empty folder with vscode
2. Run command "nanoFramework: Create New Solution" and follow instructions
3. Run command "nanoFramework: Add a Project to existing Solution" and follow instructions
4. Run command "nanoFramework: Build Project" --> fails

### Expected behaviour

Build is successfull.

### Screenshots

[image]

### Aditional context

VS Code Version: 1.92.2
OS: Manjaro Linux
Arch: x64
Nanoframework Extension Version: v1.0.181
Nanoff --version: nanoff 2.5.90+c66e018c48
msbuild --version: 16.10.1.36201

MexHigh, 2024-09-01:
**Some more info on the hanging flash command from the extension**: In the vscode output console I see a node exception with EACCESS while reading from serial. However I can read and write manually from the command line and I'm in all correct dialout permission groups. I also confirmed that vscode runs as the same user.

When building the vsix manually from source, this error does not occur and I can flash the image.

---

**Some more info on the missing build target**: I figured out, that the build variable `NanoFrameworkProjectSystemPath` passed to msbuild via property flag has incorrect casing in the path. The value of the variable is used to include more targets:

```xml
<Import Project="$(NanoFrameworkProjectSystemPath)NFProjectSystem.CSharp.targets" />
```

The extension passes `-p:NanoFrameworkProjectSystemPath=/home/leon/.vscode/extensions/nanof
ramework.vscode-nanoframework-1.0.181-g736b9e5a52/dist/utils/nanoFramework/v1.0/` to mbuild, however the path does not exists with `.../nanoFramework/v1.0`, but with `.../nanoFramework/v1.0`. I fixed the paths at the following locations:
- https://github.com/nanoframework/nf-VSCodeExtension/blob/main/src/dotnet.ts#L27
- https://github.com/nanoframework/nf-VSCodeExtension/blob/main/src/dotnet.ts#L31
- https://github.com/nanoframework/nf-VSCodeExtension/blob/main/src/dotnet.ts#L45
- https://github.com/nanoframework/nf-VSCodeExtension/blob/main/src/dotnet.ts#L46

After fixing this, I get another error message: `error CS0234: The type or namespace name 'Hardware' does not exist in the namespace 'nanoFramework' (are you missing an assembly reference?)`

I don't know what that means. Any ideas or fixes?

MexHigh, 2024-09-01:
More Info: Flashing does not work either, because neither building nor flashing does work with the `vscode-nanoframework.nfdeploy` command due to the use of child_process.exec function (I think).

Ellerbach, 2024-09-02:
@MexHigh the VS Code Extension is still an experiment. I does need some love. So, if you are willing to fix those issues, we will be very happy to get some PR!
The very very very large majority of the users are using Windows and Visual Studio with the nanoFramework associated extension. The VS Code and non Windows users are a minority. We are willing to support more but it does require motivated contributors to boost it!

MexHigh, 2024-09-03:
Ohh, I didn't knew that the extension is still experimental. It would be nice if there was some kind of warning in the Repos README or at least in the documentation.

I'll tinker with it a bit more and create a PR for it if I can get it to work.

Ellerbach, 2024-09-03:
@MexHigh any improvement, including in the README would be great! Thanks a lot!

MexHigh, 2026-02-27:
@josesimoes The build step seems to work now, thank you for working on it!

However, I am currently not able to confirm the flashing issues, since I do not have an environment to test it.