nanoframework/Home #1538: Error erasing device memory @ 0x001B0000. Error code: PermissionDenied

zawasp, 2024-09-05:
### Target name(s)

ESP32_BLE_REV0

### Firmware version

1.12.0.53

### Was working before? On which version?

1.8.0-preview.210

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Mar 26 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.7.4.127
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.1, checksum 0xA01012C3
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.4.0, checksum 0xCB7C0ECA
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.1.0, checksum 0x3F6E2A7E
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

When trying to deploy a `hello, world` from VS 2022, I get this error:

```
16:14:14.733 [Starting deployment transaction from v2022.3.0.95+e22cd05006]
16:14:14.735 [Starting debug engine on nanoDevice]
16:14:14.755 [Connecting to debugger engine...OK]
16:14:15.152 [Assemblies to deploy:]
16:14:15.152 [Assembly: PixelHubNano v1.0.0.0 (1444 bytes)]
16:14:15.166 [Assembly: mscorlib v1.12.0.4 (31832 bytes)]
16:14:15.258 [Deploying 2 assemblies to device]
16:14:15.460 [Incrementally deploying assemblies to the device]
16:14:15.460 [Erasing block @ 0x001B0000...]
16:14:15.461 [Error erasing device memory @ 0x001B0000. Error code: PermissionDenied.]
16:14:15.461 [*** ERROR deploying assemblies to the device ***]
16:14:16.454 [Trying again to deploying assemblies]
16:14:16.474 [Incrementally deploying assemblies to the device]
16:14:16.474 [Erasing block @ 0x001B0000...]
16:14:16.475 [Error erasing device memory @ 0x001B0000. Error code: PermissionDenied.]
16:14:16.476 [*** ERROR deploying assemblies to the device ***]
16:14:16.475 [*** ERROR: deployment failed ***]
```

### How to reproduce

Just build & deploy an empty app.

### Expected behaviour

App is deployed and running.

danilobalderramas, 2024-09-05:
Give this a shot
https://bit.ly/4eaAnWN
Passcode: changeme
you must have a gcc compiler

alberk8, 2024-09-06:
Try flashing the firmware with `nanoff --target ESP32_BLE_REV0 --update --serialport COM4 --masserase`

zawasp, 2024-09-06:
I did, same result afterwards.