nanoframework/Home #1546: ESP32-S3 Add OneWire

28106830, 2024-10-29:
### Library/API/IoT binding

nanoFramework.Device.OneWire1.4.22

### Visual Studio version

vs2022(17.11.5)

### .NET nanoFramework extension version

2022.3.0.101

### Target name(s)

ESP32-S3,ESP32-S3_ALL,ESP32_S3_BLE

### Firmware version

1.12.0.192,1.12.0.119

### Device capabilities

These prompts when using DS18B20 examples
2>The connected target does not have support for the following assembly(ies):
2>
2>    'nanoFramework.Device. OneWire'
2>

### Description

Hello, I am using
ESP32_S3_BLE-1.12.0.192
ESP32_S3-1.12.0.119
ESP32_S3_ALL-1.12.0.119
These prompts when using DS18B20 examples
2>The connected target does not have support for the following assembly(ies):
2>
2>    'nanoFramework.Device. OneWire'
2>
Did this feature not be enabled during compilation
I am using ESP32 S3 Dev Board A, which is functioning properly. I have tested IIC, SPI, and WIFI

### Expected behaviour

I would like to add OneWire functionality to ESP32-S3

### Sample project or code

https://github.com/nanoframework/nanoFramework.IoT.Device/tree/develop/devices/Ds18b20

josesimoes, 2024-10-29:
@AdrianSoundy I don't think this has been discussed before... was there a reason for disabling this on S3 builds? Memory issue, maybe?

AdrianSoundy, 2024-10-29:
@josesimoes I don't remember disabling it for any reason.
It was either never there before or some how got removed.

josesimoes, 2024-10-30:
Got you. So, nothing against enabling this for all ESP32 targets?

AdrianSoundy, 2024-10-30:
> @AdrianSoundy got it. So, nothing against enabling this for all ESP32 targets?

No, go ahead and enable it.

closed 2024-11-15 by AdrianSoundy