nanoframework/Home #1547: Tinyfnt resource file are always added as String

Feiko, 2024-11-06:
### Tool

Visual Studio extension

### Description

If you add a tinyfnt file to the nf resource manager it will be added as a string and not a Font. 

### How to reproduce

1. Download [small.tinyfnt](https://github.com/nanoframework/Samples/blob/ee977e8c702a4e3e37ba7bc8fcaaca25978a31ad/samples/Graphics/Primitives/Resources/small.tinyfnt)
2. Create a new nFSolution
3. Add a new "Resource File (.NET nanoFramework)"
4. Add a new resource
5. Select Type "File" and select the small.tinyfnt
6. Click save or "run custom tool"
7. The generated Recource.Desinger.cs output will only show "internal static string GetString(Resource.StringResources id)" And is missing the expected "internal static string GetFont(......)"

### Expected behaviour

The generated Recource.Desinger.cs output will only show "internal static string GetString(Resource.StringResources id)" And is missing the expected "internal static string GetFont(......)"

### Aditional context

latest clean VS2022 with nanoFramework extension

Feiko, 2024-11-08:
Looks like this issue is caused by a bug in Visual Studio. Please check [following bug report](https://developercommunity.visualstudio.com/t/Resource-Manager---Added-Files-are-alway/10785906)

josesimoes, 2024-11-12:
Waiting for fix upstream. Reportedly will be released in VS v17.12

networkfusion, 2024-11-15:
This has now been released. Hopefully fixed when using the new nF VS extension

josesimoes, 2024-11-15:
Nice catch! I did try using resources with the new version and those are working again.
The fix upstream did it.

closed 2024-11-15 by josesimoes