nanoframework/Home #1548: Panic on startup with latest firmware for Wrover-Kit

AdrianSoundy, 2024-11-07:
### Target name(s)

Wrover-kit

### Firmware version

1.12.0.119

### Was working before? On which version?

1.11.0.3

### Description

After newly flashing wrover-kit firmware its never seen in VS.

Reported on discord 
https://discord.com/channels/478725473862549535/1302792281752866816/1302961122541830244

After further testing found it panics on startup.

### How to reproduce

Flash wrover-kit firmware
reset device

### Expected behaviour

Boots and seen in VS

### Aditional information

```
[1B][0;32mI (1339) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (1346) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[1B][0m
[1B][0;32mI (1352) heap_init: At 3FFC1BC0 len 0001E440 (121 KiB): DRAM[1B][0m
[1B][0;32mI (1358) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[1B][0m
[1B][0;32mI (1365) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[1B][0m
[1B][0;32mI (1371) heap_init: At 4009B158 len 00004EA8 (19 KiB): IRAM[1B][0m
[1B][0;32mI (1378) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator[1B][0m
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x401ec050  PS      : 0x00060534  A0      : 0x80091ad8  A1      : 0x3ffe3a90  
A2      : 0x3ff42000  A3      : 0x3ffe3b1f  A4      : 0x00000000  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x8009174e  A9      : 0x00000004  
A10     : 0x3ffe3b1c  A11     : 0x3ffe3a90  A12     : 0x00000003  A13     : 0x00000000  
A14     : 0xbfffffff  A15     : 0x3ffb07b8  SAR     : 0x00000008  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40099d1a  LEND    : 0x40099d1e  LCOUNT  : 0x00000000  

Backtrace: 0x401ec04d:0x3ffe3a90 0x40091ad5:0x3ffe3ad0 0x40099d37:0x3ffe3b00 0x40088136:0x3ffe3b40 0x401a5e81:0x3ffe3b70 0x4008823a:0x3ffe3b90 0x401a5cdf:0x3ffe3bc0 0x40158cd8:0x3ffe3c20 0x40083bc5:0x3ffe3c60 0x400795ce:0x3ffe3c90 |<-CORRUPTED
```

Back trace
```
0x401ec04d: __memcpy_aux at memcpy.S:129
0x40091ad5: spi_flash_hal_common_command at spi_flash_hal_common.inc:180
0x40099d37: memspi_host_read_id_hs at memspi_host_driver.c:99
0x40088136: read_id_core at esp_flash_api.c:424
0x401a5e81: esp_flash_read_chip_id at esp_flash_api.c:442
0x4008823a: esp_flash_init_main at esp_flash_api.c:351
0x401a5cdf: esp_flash_init_default_chip at esp_flash_spi_init.c:367
0x40158cd8: do_core_init at startup.c:333
 (inlined by) start_cpu0_default at startup.c:467
0x40083bc5: call_start_cpu0 at cpu_start.c:784 (discriminator 1)

```

AdrianSoundy, 2024-11-08:
The last working firmware was just before IDF 5.1 to 5.22 upgrade. Nothing obvious in changed files to cause this.

Tested an IDF example *wifi station) with wrover and its OK.

josesimoes, 2024-11-08:
Commenting out code @ spi_flash_hal_common_command it does get past that, and tries to start both tasks on each core (at app_main), however (and not surprisingly) it fails short after with the error that the flash is not initialized.

AdrianSoundy, 2024-12-14:
I have ruled out the bootloader.bin as the cause of the problem.

Blink example built under IDF 3.5.2 works OK on Wrover_Kit
Using the bootloader.bin created from the IDF blink sample has the same problem when flashed with nanoclr.bin
Using nanoframework bootloader.bin with blink.bin app runs OK

So issue is with nanoclr.bin !

networkfusion, 2025-03-07:
@AdrianSoundy was this resolved?