nanoframework/Home #1549: Wi-Fi helper has exception if you try to re-connect

AdrianSoundy, 2024-11-10:
### Library/API/IoT binding

System.Device.Wifi

### Visual Studio version

VS2022

### Target name(s)

ESP32_S3_ALL

### Firmware version

1.12.0.119

### Description

If you have code that tries to reconnect the wifi after disconnecting or if initial connection failed you get an invalidOperation Exception.  If a device decides to disconnect wifi so as to save power and to light sleep you can not reconnect with out having to restart device.

See example code where it will disconnect device to sleep or collect data then try to reconnect.
Also if the connect fails it tries to re-connect but this also fails. 

This has been reported before but this is a bug not a feature request,

Commenting out the exception in the library does seem to make it work but this may not be the right way to fix it.

### How to reproduce

1. Run provided code.
2. When trying to reconnect will get exception

### Expected behaviour

It should allow you to retry connecting wifi without having to restart.

### Sample project or code

Example code
```
            while (true)
            {
                Console.WriteLine("Connecting WiFI network");

                CancellationTokenSource cs = new(60000);

                bool result = WifiNetworkHelper.ConnectDhcp(MySsid, MyPassword, WifiReconnectionKind.Automatic, true, 0, cs.Token);
                if (!result)
                {
                    Console.WriteLine($"{DateTime.UtcNow}  Wifi result = false");

                    // Retry in 1 minute
                    Thread.Sleep(60000);
                    continue;
                }

                Console.WriteLine($"{DateTime.UtcNow}  Wifi result = true");

                // Do work
                Thread.Sleep(20000);

                // Disconnect & sleep (light)
                WifiNetworkHelper.Disconnect();

                Thread.Sleep(5000);
            }
```

### Aditional information

The exception you get is :
```
    ++++ Exception System.InvalidOperationException - 0x00000000 (1) ++++
    ++++ Message: 
    ++++ nanoFramework.Networking.NetworkHelper::SetupHelper [IP: 0008] ++++
    ++++ nanoFramework.Networking.NetworkHelper::InternalWaitNetworkAvailable [IP: 0004] ++++
    ++++ nanoFramework.Networking.WifiNetworkHelper::ScanAndConnect [IP: 015b] ++++
    ++++ nanoFramework.Networking.WifiNetworkHelper::ConnectDhcp [IP: 000d] ++++
    ++++ WifiReconnect.Program::Main [IP: 002c] ++++
```

Ellerbach, 2024-11-12:
Duplicate. See https://github.com/nanoframework/Home/issues/1421

In short:
* It's "by design"
* You can use the following pattern to have it working: https://github.com/nanoframework/Samples/blob/4a66f23dc6b69c3d05f3a874eb2a1c38df60234f/samples/WiFiAP/Wireless80211.cs#L82

So, I will close this issue as there is a way around it.

closed 2024-11-12 by Ellerbach

Ellerbach, 2024-11-25:
> I think the design should change to allow this.

I'm fine, we had some time ago a long debate (still in the other thread) about this. If we want to change the logic of the helper, it's doable of course. It will require quite some rework based on how it's done. And still it won't cover all the cases as some cases need a reboot! So, you will have to know what you are doing.