nanoframework/Home #1550: VS extension fails to load following update to VS2022 v17.12

josesimoes, 2024-11-13:
### Tool

Visual Studio extension

### Description

On loading VS, there is a message about the extension failing to load.
The "detail" on VS activity log is useless:
` [Visual Studio 2022 extension for .NET nanoFramework. Enables creating C# Solutions to be deployed to a target board and provides debugging tools.](null)`
 

### Aditional context

Started following update to VS2022 to v17.12.
Reported to Dev Community [here]().

28106830, 2024-11-15:
Close plugin
Just enable the plugin again

28106830, 2024-11-15:
Temporary handling methods
1. First in VS2022
2. Close the Device Explorer window
3. Close VS2022
4. Just open it again

closed 2024-11-15 by josesimoes