nanoframework/Home #1551: ESP32-C3: System.Exception CLR_E_FAIL on SerialPort::Open

TheGuy920, 2024-11-14:
### Library/API/IoT binding

System.IO.Ports

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.3.0.103

### Target name(s)

XIAO_ESP32_C3

### Firmware version

1.12.1.7

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32_C3 built with ESP-IDF -128-NOTFOUND
  Target:   XIAO_ESP32C3
  Platform: ESP32

Firmware build Info:
  Date:        Nov 13 2024
  Type:        MinSizeRel build, chip rev. >= 3, without support for PSRAM
  CLR Version: 1.12.1.7
  Compiler:    GNU ARM GCC v13.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  100000000064E83384BA6C

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  SOAR, 1.0.0.0
  nanoFramework.Runtime.Events, 1.11.18.0
  System.Device.I2c, 1.1.16.0
  mscorlib, 1.15.6.0
  nanoFramework.System.Text, 1.2.54.0
  System.Buffers.Binary.BinaryPrimitives, 1.2.0.0
  Iot.Device.Ads1115, 1.2.0.0
  System.Device.Spi, 1.3.52.0
  UnitsNet.Length, 5.60.0.0
  UnitsNet.Frequency, 5.60.0.0
  UnitsNet.ElectricPotential, 5.60.0.0
  System.IO.Ports, 1.1.86.0
  System.IO.Streams, 1.1.59.0
  System.Device.Gpio, 1.1.41.0
  System.Math, 1.5.43.0
  Iot.Device.Hx711, 1.1.0.0
  nanoFramework.Runtime.Native, 1.7.1.0
  nanoFramework.Hardware.Esp32, 1.6.19.0
  System.Diagnostics.Stopwatch, 1.2.0.0
  UnitsNet.Temperature, 5.60.0.0
  Iot.Device.Max31856, 1.2.0.0

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.10.0, checksum 0x0EAE898B
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.5.0, checksum 0xB97BFDE3
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.2.0, checksum 0x40DC251F
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.3, checksum 0x343142CA
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.3, checksum 0xB0025409
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.11, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3fcc036c  0x0001c000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

### Description

I am simply trying to open COM2 and it fails. I've checked other issues in the past, although the tend to follow the expected exceptions under the documentation for `.Open();`. The error I've run into deviates from that and returns a generic `System.Exception`. I have tried no `<DefineConstants>` and I've tried `ESP32` +/ `BUIID_FOR_ESP32` +/ `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG_ENABLED` in different combinations with no luck.

```
Available COM Ports: [ COM2, ]
Opening port COM2...
    ++++ Exception System.Exception - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.IO.Ports.SerialPort::NativeInit [IP: 0000] ++++
    ++++ System.IO.Ports.SerialPort::Open [IP: 000f] ++++
Exception thrown: 'System.Exception' in System.IO.Ports.dll

An unhandled exception of type 'System.Exception' occurred in System.IO.Ports.dll
```

### How to reproduce

## Repo
- Flash firmware onto XIAO_ESPP32_C3
- New project
- Paste code into main and install required libraries
- Compile and run
- Cry
```
Configuration.SetPinFunction(Gpio.IO20, DeviceFunction.COM2_RX);
Configuration.SetPinFunction(Gpio.IO21, DeviceFunction.COM2_TX);

SerialPort port = new("COM2", 115_200);
port.Open();
```

### Expected behaviour

The serial port should open with no issue or with a more detailed and appropriate exception. 
Currently fails to do either

### Screenshots

[image]

### Sample project or code

```js
// Pin setup
Configuration.SetPinFunction(Gpio.IO20, DeviceFunction.COM2_RX);
Configuration.SetPinFunction(Gpio.IO21, DeviceFunction.COM2_TX);

// Print All Available COM Ports
Debug.Write($"Available COM Ports: [");
foreach (string port in SerialPort.GetPortNames())
    Debug.Write($"{port}, ");
Debug.WriteLine("]");

// New serial port obj
SerialPort port = new("COM2", 115_200);
Debug.WriteLine("Opening port COM2");
port.Open();

Thread.Sleep(Timeout.Infinite);
```

mikmog, 2024-11-14:
I'm running the mighty XIAO_ESP32C3 on COM2. Can give it a try with newer firmware tomorrow

alberk8, 2024-11-15:
@josesimoes,  I am trying to track down the issue but don't understand what is happening.   The call to [NativeInit](https://github.com/nanoframework/nf-interpreter/blob/2a109b04e1cd748776870e0077c4cab2b7d4d8d3/targets/ESP32/_nanoCLR/System.IO.Ports/sys_io_ser_native_System_IO_Ports_SerialPort.cpp#L850) from managed immediately causes an exception.   If I add an OS_DELAY(1) in this [line](https://github.com/nanoframework/nf-interpreter/blob/2a109b04e1cd748776870e0077c4cab2b7d4d8d3/targets/ESP32/_nanoCLR/System.IO.Ports/sys_io_ser_native_System_IO_Ports_SerialPort.cpp#L861)  then there is no exception thrown.   What gives?

josesimoes, 2024-11-15:
That's most odd... Looking at the code in NativeInit the only place that returns a Sys.Exception is at the last part when it fails to create the handler task. Could it be that there is not enough memory for that and it fails?

alberk8, 2024-11-15:
> That's most odd... Looking at the code in NativeInit the only place that returns a Sys.Exception is at the last part when it fails to create the handler task. Could it be that there is not enough memory for that and it fails?

After more debugging I find that if the is a OS_DELAY(1) before [this line](https://github.com/nanoframework/nf-interpreter/blob/2a109b04e1cd748776870e0077c4cab2b7d4d8d3/targets/ESP32/_nanoCLR/System.IO.Ports/sys_io_ser_native_System_IO_Ports_SerialPort.cpp#L895) then there is no exception and if I move it below then the exception appears.

TheGuy920, 2024-11-15:
I downgraded to `1.12.0.213` and all is working perfectly. For some reason it took an absolute genius to recommend I downgrade. I feel like I've just struggled through this issue for hours just to be hit with the "have you tried restarting" to fix it.

I am glad to hear the issue was reproduceable. I want to say that I don't think the problem is a lack of available memory but tbh I have no idea. There's a reason I am using C# on microcontrollers and its to stay away from this memory nonsense.

Feiko, 2024-11-15:
Just ran into the same issue on multiple ESP32 platform, including the S3 and esp32 PICO. I found that the code does work if you put in a Thread.Sleep or halt the code via a breakpoint for a second

SerialPort port = new SerialPort("COM3", 9600);
Thread.Sleep(1000);
port.Open(); //this will work

AdrianSoundy, 2024-11-18:
The problem for C3 and any unicore device is the code tries to start the Uart event task on the 2nd core which fails. Not sure why it worked before.  This is probably a problem on all ESP32 riscv devices. 
I will do a PR for a fix.

I tested this fix with ESP32 and it also seems to have resolved problem. As no delay is now required from my testing where before it gave error.  Please let me know if there is still a problem.

closed 2024-11-18 by AdrianSoundy