nanoframework/Home #1554: WiFi connection failed on ESP32-S3FH4R2

28106830, 2024-11-15:
### Target name(s)

ESP32-S3FH4R2  PICO

### Firmware version

ESP32_S3-1.12.1.3.zip

### Description

When using ESP32-S3FH4R2, using the wifi example, the following code freezes (tested on 4 boards)
WifiConnectionResult result = sender.Connect(net, WifiReconnectionKind.Automatic, MYPASSWORD);
There is no problem when using another motherboard: ESP32-WROOM-1
There is no problem when using another motherboard: ESP32-S3R8 chip

### How to reproduce

1. Use ESP32-S3FH4R2
2. Use example WiFi

28106830, 2024-11-18:
When executed to var rev=wifi Adapter Connect(net, WifiReconnectionKind.Automatic, WifiPassword); When using single step debugging, the output area displays the following information:
Single step execution: Non user code is being executed step by step“System.Device.Wifi.WifiAdapter.Connect”
Single step execution: Non user code is being executed step by step“System.Device.Wifi.WifiAdapter.Connect”
Single step execution: Non user code is being executed step by step“System.Device.Wifi.WifiAvailableNetwork.Ssid.get”
Single step execution: Non user code is being executed step by step“System.Device.Wifi.WifiAdapter.Conne

28106830, 2024-11-18:
Then the entire program stopped

28106830, 2024-11-21:
Has no one paid attention to this issue
This chip has a very low price and a very large usage
It is one of the most widely used chips in ESP32-S3

alberk8, 2024-11-21:
> Has no one paid attention to this issue This chip has a very low price and a very large usage It is one of the most widely used chips in ESP32-S3

Not so, the mostly it is [ESP32-S3-WROOM](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf) this is popular.

josesimoes, 2024-11-21:
@28106830 this is an Open Source project. Running on community effort. Developers work on it as time permits.
If you're in a hurry, know that there is professional support available. 
See: https://nanoframework.net/partner-program/

28106830, 2024-11-28:
It's possible that it's a problem with the circuit board. I used another board with this chip, but there's no problem with the wiring

closed 2024-11-28 by 28106830