nanoframework/Home #1555: USE STM32 F411CE

XiaoBaiXJ, 2024-11-16:
### Location

Docs website

### Description

[image]
I have already burned the firmware and connected it using a virtual serial port and USB. The virtual serial port keeps going online and offline again. When I checked with an oscilloscope, I found that the voltage was always 3.3V

### Screenshots

How should this be resolved

XiaoBaiXJ, 2024-11-18:
Do I need to use two power cords to connect the microcontroller when installing with DFU? I can use DFU to burn the framework by short circuiting 3.3V, but I cannot connect it using VS2022. Why is this?

josesimoes, 2024-11-18:
This is a question not an issue. Transfering.