nanoframework/Home #1558: Visual studio not breaking on exception

kevarek, 2024-11-21:
### Tool

Visual Studio extension

### Description

If exception is thrown, visual studio does not break and show  unhandled exception popup. Explained in thread on discord. May be related to other issue with process stack trace as explained in the thread.
https://discord.com/channels/478725473862549535/1307433046315896967/1309119132767092778

### How to reproduce

Video on discord.

kevarek, 2024-11-21:
VS2022 17.12.1; nF extension 2022.12.0.1; targets ESP32_REV0 and ESP32_REV3; ESP32_REV3-1.12.1.17.zip.

networkfusion, 2026-11-03:
@kevarek This is likely solved now. Are you able to confirm?