nanoframework/Home #1565: nanoFramework.IoT.Device dependency update process

networkfusion, 2024-12-16:
This issue type is available for Team Members only. If you are not one don't use it and choose another template appropriate to the issue you're trying to report.

### Details about Problem

nanoFramework.IoT.Device currently runs nuget dependency updates using GitHub workflows in 5 stages.
In each stage, the projects are stepped through and update sequentially (with optional exclusions).
This takes a very long time (especially stage 4)!

Once all affected projects are updated, this is re-ran by azure-pipelines before PR merge
and again after merge to main/develop.

We need to:
* work out how we can make this process as parallel as possible.
* remove duplicated builds.

Should this be done in the repo (e.g. a single workflow yaml for each solution/project)?
Should nanodu be able to parallelize the builds?

### Description

torbacz, 2024-12-16:
I've develop simple tool to keep .net pacakges up to date, because DependaBot is not working with .net well. Should be pretty easy to integrate with NF system. The key advantage is that I'm not using nuget.exe and also I've implemented in memory cache for packages thus it's pretty fast - updating 10 SLN with ~10 project each, took ~2 minutes on azure pipeline. If you are interested, I could extend it with NF and github support. 
https://github.com/torbacz/DependencyUpdated

On the other hand, there is ongoing .NET rework in DependaBot, no ETA yet.
https://github.com/dependabot/dependabot-core/issues/10865

josesimoes, 2024-12-16:
With the current workflow, I don't think we can't get past the exisiting "try to update" approach... Unless someone has a good idea on how to figure out if there is something to update. The current mechanism is already checking versions instead of performing the update, which was determined to be slightly quicker.

Having said that, the "obvious" performance upgrade to me seems to be firing parallel processes to tackle each repo.

I don't think there is a way out of the current staged run because of the interdependencies.

As for switching the the run schedule, nothing against it.

@torbacz the reason we went for a github action instead of azdo task was to take advantage of more availability simultaneously slots in github actions compared with limited azdo ones.
Still, haven't looked at your tool. Will do.
Question: anything "obvious" that you have there that could improve nanodu? 😉

torbacz, 2024-12-18:
@josesimoes 
The main differences:
- Nanodu is using nuget.exe tool, my tool is using REST API + XML parser (much faster)
- I've got InMemory Cachce for packages versions

Ellerbach, 2024-12-18:
> I've got InMemory Cachce for packages versions

you still have to build it once. Can we also expose it as a query table element? So anyone who would like to know which versions are related to which ones can have it?

torbacz, 2024-12-18:
That's right. Durring build process, so first one reaches to nuget rest api to all versions and then logic is applied (update policies and others)
https://github.com/torbacz/DependencyUpdated/blob/b98ef8e8ec96159d80fe18a1dbfa5f4be1595578/src/DependencyUpdated.Core/Updater.cs#L175

In case of IoT.Device that would be great improvement - instead of reaching to NuGet server each time, you will have to do it only once per run.

To be honset, I would not expose it. There is already info in nuspec file about it, it can be checked there for each package.

Ellerbach, 2024-12-18:
> To be honset, I would not expose it. There is already info in nuspec file about it, it can be checked there for each package.

Yes, very fair!

josesimoes, 2024-12-26:
nanodu and nanovc tools are in all our pipelines, very mature, tried and tested for a long time. As you know they are mostly wrappers around nuget CLI to serve the specificities of nanoFramework projects. I would rather improve what we have (preferably without introducing breaking changes) instead of switching to a new tool. 
This is not to dismiss, by any means, your generous offer and the work that you've done there! 😃 

Having said that, we could probably improve nanodu to use that approach of caching in memory the nuget details instead of using nuget CLI, if that's faster. 
To be honest., I was under the impression that nuget CLI would be doing something similar with querying the local nuget cache instead of keep pounding the nuget server... but I never really tested it for that.