nanoframework/Home #1569: .NET nanoFramework Extension Cannot be packaged

28106830, 2024-12-21:
### Tool

Visual Studio extension

### Description

In the package of nanoframework project properties, in the package options, set: Generate nuget at build time, even if selected, it will not be generated

### Screenshots

[image]

CoryCharlton, 2024-12-21:
Use a [.nuspec](https://github.com/CCSWE-nanoFramework/CCSWE.nanoFramework.Core/blob/master/CCSWE.nanoFramework.Core.nuspec) file and the [nuget cli](https://github.com/CCSWE-nanoFramework/actions-nanoframework/blob/7eecf0c15d28fde7db0bde91ec881fa946e6a974/.github/workflows/build-solution.yml#L58) instead.

josesimoes, 2024-12-23:
This is not an issue, rather a question. Moving to discussions.