nanoframework/Home #1582: ESP32 Build Info Not Found

alberk8, 2025-01-07:
### Target name(s)

ESP32_REV0, ESP32_REV3, ESP32_S3

### Firmware version

1.12.1.84

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF -128-NOTFOUND
  Target:   ESP32_REV0
  Platform: ESP32

Firmware build Info:
  Date:        Jan  4 2025
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.12.1.102
  Compiler:    GNU ARM GCC v13.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  1000000000246F28979F58

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.10.0, checksum 0x0EAE898B
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.5.0, checksum 0xB97BFDE3
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.2.0, checksum 0x40DC251F
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.3, checksum 0x343142CA
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.4, checksum 0x1777E2FE
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.11, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe46bc  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1B0000     nanoCLR
      1    0x001C0000       1      0x1E0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1B0000 (1728kB)   nanoCLR
  0x001C0000    0x1E0000 (1920kB)   Deployment

Deployment Map
Empty

### Description

The HAL Build Info should show the digest but only show NOTFOUND.

HAL build info: nanoCLR running @ ESP32 built with ESP-IDF -**128-NOTFOUND**

### Expected behaviour

Show the git Digest value

josesimoes, 2025-01-15:
Happens because of the way the git fetch is done. In some cases the IDF build system can't determine the tag and then outputs that wacky name. See espressif/esp-idf#10342 and related.

josesimoes, 2025-01-16:
Closed with nanoframework/nf-interpreter#3066