nanoframework/Home #1588: Wifi ScanWifi example assembly wont load

se732525, 2025-01-11:
### Library/API/IoT binding

System.Device.Wifi

### Visual Studio version

VS2022 Version 17.12.3

### .NET nanoFramework extension version

2022.12.1.14

### Target name(s)

ESP32 (ESP32-D0WD-V3 (revision v3.1))

### Firmware version

1.12.1.115

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF -128-NOTFOUND
  Target:   ESP32_REV3
  Platform: ESP32

Firmware build Info:
  Date:        Jan 10 2025
  Type:        MinSizeRel build, chip rev. >= 3, without support for PSRAM
  CLR Version: 1.12.1.115
  Compiler:    GNU ARM GCC v13.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  10000000008C4F018A1BC

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.10.0, checksum 0x0EAE898B
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.5.0, checksum 0xB97BFDE3
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.2.0, checksum 0x40DC251F
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.3, checksum 0x343142CA
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.4, checksum 0x1777E2FE
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.11, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
[... 130 lines cut ...]
> 14:50:14.170 [Attempting to connect the debugger engine (1/5)]
> 14:50:14.170 [Device running nanoCLR, updating the debug flags...]
> 14:50:14.811 [Device is running nanoCLR, requesting a restart and pause of debugger (1/5).]
> 14:50:15.421 [nanoCLR restarted successfully.]
> 14:50:15.937 [Attempting to connect the debugger engine (1/5)]
> 14:50:15.937 [Device running nanoCLR, updating the debug flags...]
> 14:50:16.594 [Device is running the CLR in initialized state]
> 14:50:16.594 [The debugger engine is listening to debug events from the nanoDevice target]
> 14:50:17.359 [Updating information about assemblies on the debugging target.]
> 14:50:18.375 [Error loading assemblies. Assemblies count is 0.]
> 

The debugger just hangs and execution never enters the Main() program. The "Error loading assemblies. Assemblies count is 0"  appears at the heart of this.

Can anyone help me with why this is happening?  The "Blinky" hello world example works just fine and debugger enters the assembly just fine.  This all happens as soon as I use nanoFramework.System.Device.Wifi which in turn requires several other dependent libraries.

I have tried the other esp32_dev3 firmware versions on nanoff. I have written my own projects like blinky and they work fine.  As soon as I try to do some wifi things and add the System.Device.Wifi nuget package it hangs up.  I have tried some lower versions of the nuget wifi package but that does not seem to matter.

### How to reproduce

Load Wifi sample from Github..  Build and deploy in the debugger.

### Expected behaviour

Should enter the Main() assembly and break on my first breakpoint.

se732525, 2025-01-11:
Sorry the assemblies from VS2022 were not attached to the description above.  Here they are:

> <packages>
> <package id="nanoFramework.CoreLibrary" version="1.15.5" targetFramework="netnano1.0" />
> <package id="nanoFramework.Runtime.Events" version="1.11.18" targetFramework="netnano1.0" /> [...]

se732525, 2025-01-11:
<packages>
<package id="nanoFramework.CoreLibrary" version="1.15.5" targetFramework="netnano1.0" />
<package id="nanoFramework.Runtime.Events" version="1.11.18" targetFramework="netnano1.0" />
<package id="nanoFramework.System.Device.Wifi" version="1.5.91" targetFramework="netnano1.0" />
<package id="nanoFramework.System.IO.Streams" version="1.1.59" targetFramework="netnano1.0" />
 <package id="nanoFramework.System.Net" version="1.11.1" targetFramework="netnano1.0" />
<package id="nanoFramework.System.Text" version="1.2.54" targetFramework="netnano1.0" />
<package id="nanoFramework.System.Threading" version="1.1.32" targetFramework="netnano1.0" />
</packages>

josesimoes, 2025-01-13:
Samples updated with latest nuget versions. Care to pull and try again? It looks like a version mismatch type of thing...

se732525, 2025-01-14:
Thank you for responding.  I loaded the sample project again and it still behaves the same.  I have also started new projects and followed the sample project steps and I still get the same behavior.  I purchased 3 of these boards and have tried them all.  Here is a link to the board: 
https://www.amazon.com/dp/B0D8S22V7L?ref_=pe_125775000_1044873430_fed_asin_title
I hope this helps.

se732525, 2025-01-14:
One more bit of information...I trued the sample bluetooth example.  I had to flash with the ESP32_BLE_REV3 platform.  The example ran fine.  So I went back to the wifi example and just added the github System.Device.Bluetooth to the wifi example but did not use any bluetooth in the code.  Bingo!  it loaded fine and debugger stopped in code and it continued on just fine.  Why did adding an unused library to the project make it work?

josesimoes, 2025-01-15:
Most likely there was a version mismatch on the nugets. When you added the BLE one, the referenced others were updated and everything worked.

I'm closing this. If you experience it again, please start a discussion. Or ask in our [Discord server](https://discord.gg/gCyBu8T).

closed 2025-01-15 by josesimoes