nanoframework/Home #1597: Сode using nanoCLR v1.12.3.11 and Iot.Device.Rtc does not work

RelaxSpirit, 2025-02-01:
### Library/API/IoT binding

Iot.Device.Rtc

### Visual Studio version

VS2022

### Description

All subsequent versions of nanoFramework NuGet packages are listed.

Testing of code using nanoCLR v1.12.3.11 and Iot.Device.Rtc does not work

When running unit tests, everything ends with an error without attempting to run the tests.

Resolving.
Link failure: some assembly references cannot be resolved!!

Assembly: Iot.Device.Rtc (1.2.0.0) needs assembly 'UnitsNet.Temperature' (5.67.0.0)
Assembly: UnitsNet.Temperature (5.67.0.0) needs assembly 'mscorlib' (1.15.6.0)
Error: a3000000
Exiting.

networkfusion, 2025-02-01:
This is due to the recent `corlib` update and `UnitsNet` nuget not being up-to-date. See https://discord.com/channels/478725473862549535/481780524864503828/1334507682152841278

[image]

RelaxSpirit, 2025-02-03:
Maybe there is a way to resolve the assembly conflict with the settings? Does nanoFramework have a BindingRedirects mechanism?
I understand that you can solve this problem simply by updating the third-party UnitsNet.Temperature is rebuilding its NuGet package with a new version of nanoFramework.CoreLibrary.

It is not possible to solve the problem on our own, the general downgrade of the NuGet nanoFramework packages in the solution only leads to checksum mismatch errors. It turns out to be a vicious circle :(

***********************************************************************
*                                                                     *
* ERROR!!!!  Firmware version does not match managed code version!!!! *
*                                                                     *
*                                                                     *
* Invalid native checksum: mscorlib 0x445C7AF9!=0xE76AC236 *
*                                                                     *
***********************************************************************

networkfusion, 2025-02-03:
Binding redirects are not yet possible in nF.

The workaround would be to revert the firmware to a working version that matches the nugets, e.g.

`nanoff --target M5Core2 --update --serialport COM7 --masserase --fwversion 1.12.1.61` with the `fwversion` being one relased before the change happened.

networkfusion, 2025-02-03:
The trick I use is (when having the sourcecode managed through version control, such as a github repo project) having a comment on the last known working (tested) FW version in the main program. When updating nugets, I update the FW to latest and install the nugets through VS.

If there is any dissagreement, I dont push.

RelaxSpirit, 2025-02-03:
Thanks for the lightning response :)

Is there any way to force nanoFramework to start.Is TestFramework also using the required version?

All my attempts lead to the fact that the latest version of nanoCLR CLI v1.0.723+cb91c37acf is taken anyway.
I tried changing the CLRVersion and/or PathToLocalCLRInstance parameters. This did not lead to clear results :(

RelaxSpirit, 2025-02-04:
After updating to the latest versions of NuGet packages, there is a version mismatch again. 
Now you need to update Iot.Device.Rtc for using the updated one UnitsNet.Temperature version 5.69.0, аt the moment, only this version uses nanoFramework.CoreLibrary version 1.16.11.

Loading nanoCLR v1.12.3.11
...Starting...
.................
Assembly: Iot.Device.Rtc (1.2.0.0)   (840 RAM - 6300 ROM - 2244 METADATA) @ 0x0000014ff6f7f844
..................
Assembly: mscorlib (1.16.11.0)   (4352 RAM - 33576 ROM - 20239 METADATA) @ 0x0000014ff6f7bc1c
....................
Assembly: nanoFramework.UnitTestLauncher (0.0.0.0)   (404 RAM - 1556 ROM - 571 METADATA) @ 0x0000014ff6f795fc
.....................

Loading Deployment Assemblies.
Resolving.
Link failure: some assembly references cannot be resolved!!
......
Assembly: Iot.Device.Rtc (1.2.0.0) needs assembly 'UnitsNet.Temperature' (5.67.0.0)
.......
Error: a3000000
Exiting.

josesimoes, 2025-02-04:
There a bunch of cascade updates on all the bindings. The process was triggered a couple of hours ago, is still running.
Please be more patient. 
For future reference, you can see if the pipeline is running by looking at the icon in the repo:

[image]

stephan-lechner, 2025-02-04:
Did the nfbot get rate-limited here? https://github.com/nanoframework/nanoFramework.M5Stack/pull/380

josesimoes, 2025-02-04:
@stephan-lechner nope, some nugets still not bumped to the latest versions

[image]

RelaxSpirit, 2025-02-05:
After updating the NuGet nanoFramework.Iot.Device.Rtc version 1.2.775 the problem has disappeared.
Thank you all for your participation.
Can I close this problem?

josesimoes, 2025-02-05:
Yes please. Thank you for the positive feedback.