nanoframework/Home #1601: New firmware 1.12.3.x thrown a System.InvalidOperationException at the I2C channel.

romankiss, 2025-02-11:
### Target name(s)

S3, C6, C3

### Firmware version

1.12.3.x

### Was working before? On which version?

1.12.1.x

### Description

```C#
Configuration.SetPinFunction(pinI2C1_SDA, DeviceFunction.I2C1_DATA);  
Configuration.SetPinFunction(pinI2C1_SCL, DeviceFunction.I2C1_CLOCK); 
bool IsHATused = false;
// I2C scanner
for (int ii = 0; ii < 255; ii++)
{
    var i2csettings = new I2cConnectionSettings(1, ii);
    var result = I2cDevice.Create(i2csettings).WriteByte(0x07);      // !!! thrown Exception at ii=0x7B
    if (result.Status != I2cTransferStatus.FullTransfer) continue;
    IsHATused = true;
    Debug.WriteLine($">>> The First I2C Device on BusId={i2csettings.BusId} found it, address=0x{ii:X}");
    break;
}
```

The following Exception is thrown:

```C#
Exception System.InvalidOperationException - CLR_E_INVALID_OPERATION (1) 
Message: 
System.Device.I2c.I2cDevice::NativeTransmit [IP: 0000] 
System.Device.I2c.I2cDevice::WriteByte [IP: 001f] 
Exception thrown: 'System.InvalidOperationException' in System.Device.I2c.dll
An unhandled exception of type 'System.InvalidOperationException' occurred in System.Device.I2c.dll
```

The exception is thrown for either the empty (no device) or connected device on the I2C Bus. When I made a rollback to the firmware version for instance 1.12.1.47 on my target ESP32_S3, everything is working. 

Notice, that when the I2C scanner code is commented, everything is working well. Another symptom happens at the target ESP32_S3_BLE and ESP32_S3_ALL, where the I2C scanner has been passed but this _System.InvalidOperationException_ has been thrown at the I2C device, for instance Vl53L0X.

t looks like something is going wrong at the **NativeTransmit** level after ESP-IDF has been updated. 
Also, the interesting is why the I2C scanner thrown an Exception at the address 0x7B?

muenchris, 2025-02-13:
A quick Side note:
```
var result = I2cDevice.Create(i2csettings).WriteByte(0x07); 
```
is dangerous as there are some I2C devices that will do harmful things when you write to them.
Especially Relay Boards like this one: https://www.tindie.com/products/bugrovs2012/8-channel-i2c-electromagnetic-relay-module/

muenchris, 2025-02-13:
> Adding a speculative note that `nanoFramework.System.Runtime` was changed and for the most part is now included in corlib. Removing that nuget package from your local project "may" fix the issues.

I have tried this, but still get the error after several successful writes. This is consistent with the report from @romankiss above. It looks like many writes succeed but at some point it fails - hence the weird start of failure at address 0x7B.

josesimoes, 2025-02-14:
Tested on ORGPAL_THREE for cross checking with another platform (STM32). 
Successfully run I2C scanner (from samples repo) on both I2C2 and I2C3. 

```text
Attaching to nanoDevice...
Waiting for nanoDevice to initialize...
Debugger found. Resuming boot sequence.
Create Type System.
Loading Deployment Assemblies.
   Assembly: NanoI2cScanner (1.0.0.0)   (264 RAM - 996 ROM - 345 METADATA)

   AssemblyRef    =        8 bytes (       2 elements)
   TypeRef        =       48 bytes (      12 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       52 bytes (      13 elements)
   TypeDef        =        8 bytes (       1 elements)
   FieldDef       =        0 bytes (       0 elements)
   MethodDef      =        4 bytes (       2 elements)
   StaticFields   =        0 bytes (       0 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =      303 bytes
   Signatures      =       65 bytes
   ByteCode        =      220 bytes

   Assembly: nanoFramework.Runtime.Events (1.11.29.0)   (596 RAM - 3412 ROM - 1799 METADATA)

   AssemblyRef    =        4 bytes (       1 elements)
   TypeRef        =       48 bytes (      12 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       48 bytes (      12 elements)
   TypeDef        =      120 bytes (      15 elements)
   FieldDef       =       44 bytes (      22 elements)
   MethodDef      =      104 bytes (      52 elements)
   StaticFields   =       36 bytes (       3 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =      739 bytes
   Signatures      =      227 bytes
   ByteCode        =      747 bytes

   Assembly: System.Device.I2c (1.1.26.0)   (340 RAM - 1756 ROM - 826 METADATA)

   AssemblyRef    =        4 bytes (       1 elements)
   TypeRef        =       32 bytes (       8 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       28 bytes (       7 elements)
   TypeDef        =       40 bytes (       5 elements)
   FieldDef       =       24 bytes (      11 elements)
   MethodDef      =       48 bytes (      23 elements)
   StaticFields   =        0 bytes (       0 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =      379 bytes
   Signatures      =      118 bytes
   ByteCode        =      424 bytes

   Assembly: mscorlib (1.16.11.0)   (4156 RAM - 33576 ROM - 20239 METADATA)
[... 45 lines cut ...]
'nanoFramework.Tools.VS2022.Extension.dll' (Managed): Loaded 'E:\GitHub\nf-Samples\samples\I2C\NanoI2cScanner\packages\nanoFramework.CoreLibrary.1.16.11\lib\mscorlib.dll'
'nanoFramework.Tools.VS2022.Extension.dll' (Managed): Loaded 'E:\GitHub\nf-Samples\samples\I2C\NanoI2cScanner\bin\Debug\NanoI2cScanner.exe', Symbols loaded.
'nanoFramework.Tools.VS2022.Extension.dll' (Managed): Loaded 'E:\GitHub\nf-Samples\samples\I2C\NanoI2cScanner\packages\nanoFramework.Runtime.Events.1.11.29\lib\nanoFramework.Runtime.Events.dll'
'nanoFramework.Tools.VS2022.Extension.dll' (Managed): Loaded 'E:\GitHub\nf-Samples\samples\I2C\NanoI2cScanner\packages\nanoFramework.System.Device.I2c.1.1.26\lib\System.Device.I2c.dll'
The thread '<No Name>' (0x2) has exited with code 0 (0x0).
Hello from I2C Scanner!
0x00 Write: 4, transferred: 0, Read: 4, transferred: 0
0x00 - Absent
0x01 Write: 4, transferred: 0, Read: 4, transferred: 0
0x01 - Absent
0x02 Write: 4, transferred: 0, Read: 4, transferred: 0
0x02 - Absent
(.........)
0xFC Write: 4, transferred: 0, Read: 4, transferred: 0
0xFC - Absent
0xFD Write: 4, transferred: 0, Read: 4, transferred: 0
0xFD - Absent
0xFE Write: 4, transferred: 0, Read: 4, transferred: 0
0xFE - Absent
0xFF Write: 4, transferred: 0, Read: 4, transferred: 0
0xFF - Absent
```

This seems to validate that there are no issues with I2C C# library and native layer on a different platform. 
Pointing out to an issue specific to ESP32 devices.

Ellerbach, 2025-02-14:
It seems that the problem appear after "quite some time" like "10 minutes". So, @josesimoes try to run the code for 1 hour or so and same for @torbacz and see how it behaves.
If both of your tests behaves properly, then, it's most likely something else that has a side effect on the I2C, and that will be harder to track.

muenchris, 2025-02-14:
> It seems that the problem appear after "quite some time" like "10 minutes". So, [@josesimoes](https://github.com/josesimoes) try to run the code for 1 hour or so and same for [...]

I have about 5 sensors that I read every second on my OLIMEX and they read all fine without issues for about 10 cycles, then suddenly only some of the sensors fail. (as seen in my post above). The other sensors continue to read correctly. With 1.12.1 I could read those sensors for days without any issues.
The odd thing is that its always the same sensors. For Example in the BME280, it still reads the Pressure and Humidity just fine, but the temperature and the Altitude fails. On the ADS1115 Pin 0 works but Pin 1 fails.

josesimoes, 2025-02-14:
@muenchris are those on the same I2C bus? Do you create the I2C device on each usage or once and then reuse it?

muenchris, 2025-02-14:
> [@muenchris](https://github.com/muenchris) are those on the same I2C bus? Do you create the I2C device on each usage or once and then reuse it?

Yes, same I2C Bus. Using Bus1 on Pins 2 and 3 on the OLIMEX.
As for the usage, on the BME I reuse an opened I2C Device and on the ADS1115 I create a new I2C device for each read - even for each Pin Read.

josesimoes, 2025-02-14:
Got you. Performance wise, for objects being used over and over, it's always preferable to create them once and reuse when needed. This saves potential GC runs.

muenchris, 2025-02-14:
> Got you. Performance wise, for objects being used over and over, it's always preferable to create them once and reuse when needed. This saves potential GC runs.

Yes, this is my plan for tuning the system as long as there is enough memory available to store all my objects.

torbacz, 2025-02-18:
Just finished 1hr loop - 3 sensors connected, on the same I2C bus. The bus and devices are created once, then only communication. All works flawsless - no errors.

muenchris, 2025-02-18:
What speeds are you using for your devices? I tried with standard, fast and superfast and get the same results. 
For example the BME280 reads humidity and pressure just fine but fails to read temp and altitude. When I start the application it runs for about 10-15 cycles without problems including temp and alt, but then fails always at these two readings.
I also create the bus and devices once and just communicate with them. 

In my current test setup I use these sensors:
ADS1115, BME280, INA219, MCP23017 
I also use the neopixel library for the WS2812B with 29 LEDs.
All sensors are read every second.

This setup works flawlessly with 1.12.1.199. If I upgrade to 1.12.3 it runs fine for 10-15 cycles then the ADS1115 and BME280 fails to read and the MCP23017 fails to write.

The app has a websocket connection open to a host controller. It can send commands to the ESP which are handled on a second thread. This again works just fine on 1.12.1 and 1.12.3. But I am wondering if there might be a collision/timing issue in 1.12.3 that throws the I2C bus off. 
Since NF does not have CriticalSections I am not sure if the mscorlib is protecting the I2C Bus traffic from Thread interruptions. 
Did this behavior maybe change in 1.12.3?

romankiss, 2025-02-19:
@torbacz , try the following (on the one of S3/C3/C6 target):

1. Create the i2c Bus
2. Run the i2c Scanner (see the above code)
3. Create the i2c Devices (once)
4. Run a communication for devices in the timer/button handler.

josesimoes, 2025-02-19:
@muenchris I2C communication occurs in IDF code. We're "just" calling the IDF API. The Threads execution doesn't interrupt it.

AdrianSoundy, 2025-02-22:
This is a problem related to dispose being called by GC. Its an error in the driver which i will fix shortly.

That is why it works if you only open device once. If you device is created and destroyed the problem will occur. There maybe a work around by implicitly disposing IC2device object after each use.

closed 2025-02-22 by AdrianSoundy

romankiss, 2025-02-22:
@AdrianSoundy, Thanks for resolving this issue. I can confirm that the newest version such as **1.12.3.104** on the target **ESP32_S3_ALL** and **ESP32_C6_THREAD** is working well with my I2C Scanner and either none or connected i2c devices.

I hope that the next version of the 1.12.3.120 for other targets such as XIAO_ESP32C3, ESP32_S3 and ESP32_S3_BLE will have included this fix.

Also, thanks for improving the I2C Scanner, I am going to include it to my all ipplications.

muenchris, 2025-02-22:
just remember, if you are doing a 0x07 "random" write to a I2C device there might be unexpected consequences. Like with a Relay board that suddenly toggles relay states.

josesimoes, 2025-02-24:
@muenchris good point! The code should be using a read instead.

muenchris, 2025-02-24:
I can confirm that the fix worked in my environment, too.