nanoframework/Home #1602: ESP32_S3 debugging program doesn't start

Sandoun, 2025-02-21:
### Target name(s)

ESP32_S3

### Firmware version

1.12.3.114

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32_S3 built with ESP-IDF v5.2.3
  Target:   ESP32_S3
  Platform: ESP32

Firmware build Info:
  Date:        Feb 21 2025
  Type:        MinSizeRel build, chip rev. >= 0, support for PSRAM
  CLR Version: 1.12.3.114
  Compiler:    GNU ARM GCC v13.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  1000000000E4B323E13CF8

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  WaterIO.Firmware, 1.0.0.0
  mscorlib, 1.16.11.0

Native Assemblies:
  mscorlib v100.5.0.22, checksum 0xE76AC236
  nanoFramework.Runtime.Native v100.0.10.0, checksum 0x0EAE898B
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.5.0, checksum 0xB97BFDE3
  nanoFramework.Graphics v100.0.0.10, checksum 0x6A3F90DD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.2.0, checksum 0x40DC251F
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.3, checksum 0x343142CA
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.4, checksum 0x1777E2FE
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.11, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3fcb8e78  0x0001f008
  FLASH 0x00000000  0x00800000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x2E0000     Deployment
      2    0x00490000       1      0x200000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00490000    0x200000 (2048kB)   Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x2E0000 (2944kB)   Deployment

Deployment Map
Empty

### Description

When uploading the default example project via VS2022, the CLR doesn't seem to execute the program, it just waits for the program to start. When the ESP gets rebooted, the previously uploaded program gets executed. This was tested by attaching a serial monitor and also running a minimal led blink program and observing, that the led wont blink when uploading from VS22 but after a ESP reboot it works. 

### How to reproduce

1. Flash the esp32-s3 firmware
2.  Upload the default project

### Expected behaviour

The uploaded program should be executed

### Aditional information

Example code used:

```C#
    public class Program
    {
        public static void Main()
        {

            while (true) {

                Debug.WriteLine("Hello from nanoFramework!");

                Thread.Sleep(1000);

            }  

        }
    }
```

The esp model used is a 'ESP32-S3-WROOM-1-N8' that works totally fine with all other variants of programming it.

josesimoes, 2026-09-24:
@Sandoun one last attempt to get your feedback on this before closing the issue. Appreciate if you let us know.

josesimoes, 2026-11-02:
We are now at 1.18. This issue has almost 2 years.
No further feedback from OP so closing it.
Feel free to reopen if you are still seeing it. Better ask for support on our Discord server (https://discord.gg/gCyBu8T) as you'll get a speddier answer.

closed 2026-11-02 by josesimoes