nanoframework/Home #1604: Unable to add image to resource

TerryFogg, 2025-02-24:
### Tool

Visual Studio extension

### Description

Visual Studio Extension 2022.12.1.29
Unable to add jpeg image.
The image appears in the UI view of the resource tool but no code is added to the resource class to support it.
Fonts and strings continue to work.
Starting from a new project may allow you to add an image and get the code created, but adding a reference to the resource manager seems to stop it. Further adding causes the visual studio to add a reference to System.Drawing which is incorrect.
Microsoft have added a new resource explorer which appears to be a dogs breakfast of a UI and code, so I guess the bright spark who thought this up decided to change code that breaks exisiting code.,

networkfusion, 2025-03-07:
@TerryFogg is this the same situation if using VS2022.13.x.x with the latest extension? and do you know which VS version and extension where it last worked?

TerryFogg, 2025-03-07:
I will check

mikmog, 2025-03-15:
Encountered this as well with latest VS 2022 (17.13.3). The last resource added did not get generated to the designer file. Easy workaround though. Just add a string resource the last thing you do and the other resources will be generated. It can be removed afterwards