nanoframework/Home #1607: Bluetooh Library broken in the latest release

alberk8, 2025-03-09:
### Library/API/IoT binding

nanoFramework.Device.Bluetooth

### Visual Studio version

VS2022 17.13.2

### .NET nanoFramework extension version

2022.12.1.17

### Target name(s)

ESP32

### Firmware version

1.12.4.26

### Device capabilities

 

### Description

Running the BluetoothLESample1 encounter the error when the program execute

 ```
   ++++ Exception System.NullReferenceException - CLR_E_NULL_REFERENCE (2) ++++
    ++++ Message: 
    ++++ System.Guid::TryParseGuidWithDashes [IP: 0075] ++++
    ++++ System.Guid::.ctor [IP: 000c] ++++
    ++++ nanoFramework.Device.Bluetooth.Utilities::.cctor [IP: 0006] ++++
    ++++ Exception System.ArgumentException - 0x00000000 (2) ++++
    ++++ Message: 
    ++++ System.Guid::.ctor [IP: 0011] ++++
    ++++ nanoFramework.Device.Bluetooth.Utilities::.cctor [IP: 0006] ++++
An unhandled exception of type 'System.ArgumentException' occurred in mscorlib.dll
```

### How to reproduce

1. Download the [sample code ](https://github.com/nanoframework/Samples)
2. Compile the BluetoothLESample1 and deploy

### Expected behaviour

No error.

josesimoes, 2025-03-10:
@alberk8 make sure to update VS extension if you haven't done so.

alberk8, 2025-03-11:
Tested with new CorLibrary v1.17.7,  latest bluetooth nuget and now it is working

closed 2025-03-11 by josesimoes