nanoframework/Home #1618: Can Not Add Header To HttpClient

hamiddd1980, 2025-03-27:
### Library/API/IoT binding

nanoFramework.System.Net.Http

### Visual Studio version

Microsoft Visual Studio Enterprise 2022 (64-bit) - Preview Version 17.14.0 Preview 2.0

### .NET nanoFramework extension version

2022.12.136

### Target name(s)

ESP32-S3

### Device capabilities

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

### Description

It is not possible to add **Http** headers (like User-Agent, Accept) to **HttpClient** by **DefaultRequestHeaders** or **HttpRequestMessage** . Trying to add standard Headers, produces the following exception:
_Cannot update restricted header User-Agent.
Cannot update restricted header Accept._
Inspecting the online source code reveals that the problem arises from **WebHeaderCollection** class in **ThrowOnRestrictedHeader** method:
the class has a property named **HInfo** with type of **HeaderInfoTable** . **HeaderInfoTable**  has a table with predefined Http Headers (Like User-Agent) .
User can not add a Http Header _if that header exists in the HeaderInfoTable_.
So, how can I add (or change) standard Http Header to httpClient (Like User-Agent,Accept)?

### How to reproduce

1.Create a Static HttpClient
`httpClient = new HttpClient();`
2.Try to Add Headers To DefaultRequestHeaders
```
httpClient.DefaultRequestHeaders.Add("Accept", "application/json");
httpClient.DefaultRequestHeaders.Add("Accept-Language", "en-US,en;q=0.5");
 httpClient.DefaultRequestHeaders.Add("User-Agent", "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:136.0) Gecko/20100101 Firefox/136.0");
```

### Expected behaviour

It should be possible to Add or Change Standard Http Headers to HttpClient

Ellerbach, 2025-03-27:
As far as I remember, it's the same behavior on real .NET .Just try to add a header normally, it should just override the default one.

hamiddd1980, 2025-03-27:
I'm sure it is possible to add **Headers**(standard or user defined) to **HttpClient** in **.net Core**.Indeed, many servers doesn't accept Http request in case the headers are empty.

hamiddd1980, 2025-03-31:
Hi, is there any one here to help me?
Thanks in advance for your helps with this issue.

hamiddd1980, 2025-03-31:
I think the following line from **HttpHeaders** class should be corrected (change boolean value from **true** to **false**):
```
public abstract class HttpHeaders
    {
       internal WebHeaderCollection _headerStore = new WebHeaderCollection(true); //we should change true param to false
       //the first parameter is m_IsHttpWebHeaderObject and according to comments for this property in WebHeaderCollection class, it should be TRUE if the object is created for internal use
       

        /// <summary>
        /// true if this object is created for internal use, in this case
        /// we turn on checking when adding special headers.
        /// </summary>
        private bool m_IsHttpWebHeaderObject = false;
```

hamiddd1980, 2025-04-01:
@josesimoes , I tested the above method and regarding the results below it seems this modification can solve the problem .

**1-before modification :**
  1-1)if you try to add headers to Httpclient, you'll face the Exception :

[image]

   1-2)and if you don't add headers , you don't have headers on server side:

[image]

**2-after code modification :**
  2-1)you can include headers in Httpclient without problem:

[image]

  2-2)on the server side,you receive these headers :

[image]

I'm not familiar with Github, so it would be grateful if you could apply this Modification .
Thank you.

josesimoes, 2025-04-02:
Hi @hamiddd1980!
Thank you for the detailed investigation. Adding this to the work queue.

It's quite easy to do this: you can edit the file right on github and it will offer to open a pull request for you. No need to install anything, not clone locally.

networkfusion, 2025-04-12:
@hamiddd1980, please let me know if you can test https://github.com/nanoframework/System.Net.Http/pull/477 (and what you expected it to be).