nanoframework/Home #1625: Fix the specification of the SSL version

rikkreeftenberg, 2025-04-11:
### Library/API/IoT binding

ATModem

### Visual Studio version

VS2022

### Target name(s)

ESP32

### Firmware version

CLR Version: 1.12.3.126

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v5.2.3
  Target:   ESP32_BLE_REV3
  Platform: ESP32

Firmware build Info:
  Date:        Feb 25 2025
  Type:        MinSizeRel build, chip rev. >= 3, support for PSRAM, support for BLE
  CLR Version: 1.12.3.126
  Compiler:    GNU ARM GCC v13.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  1000000000C049EF1FF88

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  AtModem.Sample, 0.0.0.0
  System.IO.Ports, 1.1.119.0
  nanoFramework.M2Mqtt.Core, 5.1.176.0
  System.Net, 1.11.27.0
  UnitsNet.ElectricPotential, 5.69.0.0
  System.IO.Streams, 1.1.86.0
  UnitsNet.Angle, 5.69.0.0
  nanoFramework.System.Collections, 1.5.59.0
  nanoFramework.System.Text, 1.3.16.0
  nanoFramework.Runtime.Native, 1.7.8.0
  mscorlib, 1.16.11.0
  System.Math, 1.5.90.0
  nanoFramework.Runtime.Events, 1.11.29.0
  System.Threading, 1.1.46.1709
  Iot.Device.Common.GnssDevice, 1.0.0.0
  nanoFramework.Hardware.Esp32, 1.6.29.0
  UnitsNet.Speed, 5.69.0.0
  Iot.Device.AtModem, 1.0.0.0
  UnitsNet.Ratio, 5.69.0.0

Native Assemblies:
  mscorlib v100.5.0.22, checksum 0xE76AC236
  nanoFramework.Runtime.Native v100.0.10.0, checksum 0x0EAE898B
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.5.0, checksum 0xB97BFDE3
  nanoFramework.Device.Bluetooth v100.0.5.0, checksum 0xAA4B0676
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.2.0, checksum 0x40DC251F
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.3, checksum 0x343142CA
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.4, checksum 0x1777E2FE
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.11, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3f80087c  0x003b0000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1B0000     nanoCLR
      1    0x001C0000       1      0x1E0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1B0000 (1728kB)   nanoCLR
  0x001C0000    0x1E0000 (1920kB)   Deployment

Deployment Map
Empty

### Description

My LilyGO TTGO T-SIM-A7670E R2 ESP32 - ESP32-WROVER board couldn't connect to my Azure WebAPI anymore on 27-03-2025
I raised this issue with LilyGo: https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/243

Other people also had this problem when connecting to Azure Web API's.

An httpget to the url: https://ikfu.azurewebsites.net/api/GetUtcTime/
resulted in an errorcode 715. Documentation says this is a "handshake failed".

It turned out Azure disabled TLS negotiation around that time.
So you have to specify it explicitly: 
`AT+CSSLCFG="sslversion",0,3`
 
The ATModem library implements the method "httpClient.Get(...)"
When it's a https url it send the following to the modem:
`AT+CSSLCFG="sslversion",2,3`

The "2" is the SSL Context ID. 
The context ID that is used during the http request is not explicitly set. So I assume it is "0".

In the Iot.Device.AtModem.Http.Sim7672HttpClient this contextID is determined by the variable "IndexSSL" that is initialized to "2".

The solution is to set it to "0".

### How to reproduce

Connect a A7670E to an ESP32 board.
Call httpClient.Get("https://ikfu.azurewebsites.net/api/GetUtcTime/"); // Or another Azure Web API
The errorcode 715 is returned. Handshake failed.

When you call https://www.ellerbach.net/DateHeure/  everything is OK. 

### Expected behaviour

The call to httpClient.Get("https://ikfu.azurewebsites.net/api/GetUtcTime/");
should return http status 200, and a string with the UTC time.

### Sample project or code

https://github.com/nanoframework/nanoFramework.IoT.Device

nanoFramework.IoT.Device\devices\AtModem\samples\Program.cs

josesimoes, 2025-04-11:
@rikkreeftenberg this seems like a very simple fix. 
Can you go and submit a PR with it? 😉

rikkreeftenberg, 2025-04-11:
Of course: https://github.com/nanoframework/nanoFramework.IoT.Device/pull/1386

closed 2025-04-16 by Ellerbach