nanoframework/Home #1638: M5Stack Core2 v1.1 with M5Core2 nanoframework firmware hangs in nanoFramework.M5Stack.M5Core2 static class constructor

alexmiby, 2025-05-24:
### Library/API/IoT binding

nanoFramework.M5Core2

### Visual Studio version

VS2022

### .NET nanoFramework extension version

1.1.280 -- the most recent one

### Target name(s)

M5Core2

### Firmware version

The most recent one

### Description

M5Stack Core2 v1.1

alexmiby, 2025-05-24:
The hanging up occurs in nanoFramework.M5Core2 static class constructor in this line:
```c#
            // Set ADC sample rate to 25Hz
            _power.AdcFrequency = AdcFrequency.Frequency25Hz;
```
So this happens upon any program start whenever M5Core2 nuget is added, before any user code whatsoever is executed.

After the hangup, the only possible course of action is to erase firmware and flash it again using nanoff.
The source file is nanoFramework.M5Stack\nanoFramework.M5Stack\M5StackCommon\Core2ToughCommon.cs

alberk8, 2025-05-24:
@AdrianSoundy,  Could this be an issue with ADC native driver.

josesimoes, 2025-06-02:
@alexmiby if you comment out that line, does the remaining code gets executed?

alexmiby, 2025-06-06:
I tried to put the other values of AdcFrequency, and then it goes past that line further. It's specifically setting the _power.AdcFrequency to AdcFrequency.Frequency25Hz that hangs it up straight away.

josesimoes, 2026-09-24:
@alexmiby have you tried this again? There was a fix with I2C that most likely solved this. Waiting on your feedback.

josesimoes, 2026-11-02:
Closing as no feeback provided by OP.
Feel free to reopen if this is still an issue.

closed 2026-11-02 by josesimoes