nanoframework/Home #1644: Error E4000: Error executing esptool command. (The handle is invalid.)

wvdvegt, 2025-06-06:
### Tool

nanoff

### Description

I'm trying to flash my ESP32-CAM and get a 

Error E4000: Error executing esptool command. (The handle is invalid.)

as result of the 'nanoff --platform esp32 --serialport COM9 --update' command (after a lot of white chars with a black dot (directly after the progress of the backup to reach 100%).

The output (of an earlier experiment) looks like:

```
PM> nanoff --target ESP32_WROOM_32 --serialport COM6 --update --baud 115200 -v diag
.NET nanoFramework Firmware Flasher v2.5.126+716b21797a
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Using COM6 @ 115200 baud to connect to ESP32.
Reading details from chip...
Executing esptool with the following parameters:
'--port COM6  --chip auto   --after no_reset_stub flash_id'

OK                                                                                                            

Connected to:
ESP32 (ESP32-D0WD-V3 (revision v3.1))
Features WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Flash size 4MB unknown from  (manufacturer 0x104 device 0x16406)
PSRAM: 0MB
Crystal 40MHz
MAC C8:2E:18:25:A2:14

Trying to find ESP32_WROOM_32 in stable repository...OK

******************************* WARNING ******************************
** This firmware package was released at 25-8-2021                 **
** The target it's probably outdated.                               **
** Please check the current target names here: https://git.io/JyfuI **
**********************************************************************

Extracting ESP32_WROOM_32-1.7.0.2.zip...OK

Updating to 1.7.0.2

Flashing firmware...

Executing esptool with the following parameters:
'--port COM6 --baud 115200 --chip esp32   --after hard_reset write_flash --flash_size 4MB 0x1000 "C:\Users\Wim van der Vegt\.nanoFramework\fw_cache\ESP32_WROOM_32\bootloader.bin" 0x10000 "C:\Users\Wim van der Vegt\.nanoFramework\fw_cache\ESP32_WROOM_32\nanoCLR.bin" 0x8000 "C:\Users\Wim van der Vegt\.nanoFramework\fw_cache\ESP32_WROOM_32\partitions_4mb.bin"'

Uploading stub...                                                                                             
Running stub...                                                                                               
Stub running...                                                                                               
Configuring flash size...                                                                                     
Flash will be erased from 0x00001000 to 0x00007fff...                                                         
Flash will be erased from 0x00010000 to 0x0013afff...                                                         
Flash will be erased from 0x00008000 to 0x00008fff...                                                         
SHA digest in image updated                                                                                   
Compressed 25840 bytes to 15353...                                                                            
Writing at 0x00001000... (100 %)                                                                              
Wrote 25840 bytes (15353 compressed) at 0x00001000 in 1.7 seconds (effective 119.3 kbit/s)...                 
Hash of data verified.                                                                                        
Compressed 1224496 bytes to 766585...                                                                         
Writing at 0x00010000... (2 %)                                                                                
Writing at 0x0001bf58... (4 %)                                                                                
Writing at 0x00030e5a... (6 %)                                                                                
Writing at 0x0003b513... (8 %)                                                                                
Writing at 0x00043567... (10 %)                                                                               
Writing at 0x0004bc28... (12 %)                                                                               
[... 46 lines cut ...]
Hash of data verified.                                                                                        
                                                                                                              
Leaving...                                                                                                    
Hard resetting via RTS pin...                                                                                 
Error E4000: Error executing esptool command. (The handle is invalid.)
PM> 
```
If I use a command as:
 
`nanoff --platform ESP32 --target ESP32_BLE_REV3 --update --serialport COM11 --masserase -v diag`

(so with --masserase) I get a similar output (including the error message) but after reconnecting the device it shows in the device explorer.

### How to reproduce

See above for commands.

### Expected behaviour

No error message.

### Aditional context

VS2022 Pro
.NET nanoFramework Firmware Flasher v2.5.126+716b21797a

josesimoes, 2025-06-06:
You may have noticed the warning there: 

```

******************************* WARNING ******************************
** This firmware package was released at 25-8-2021                 **
** The target it's probably outdated.                               **
** Please check the current target names here: https://git.io/JyfuI **
**********************************************************************

```

The target `ESP32_WROOM_32` was deprecated in 2021. You should choose another one appropriate to your hardware. In doubt replace `--target NNNAAAMMMEEE` with `--platform esp32` and the tool will choose one for you.

wvdvegt, 2025-06-06:
Please re-open, it's not limited to the Wroom. I''ve updated the output.