nanoframework/Home #1653: WebServer Running HTTPS

renerlemes, 2025-06-27:
### Library/API/IoT binding

nanoFramework.WebServer

### Target name(s)

ESP32_S3_ALL

### Firmware version

Latest

### Description

When I'm using WebServer on port 80 with HTTP, everything works normally, but when I use the example itself that is in the repository (https://github.com/nanoframework/Samples/blob/main/samples/Webserver/WebServer.Sample/WebServer.Sample/Program.cs), I receive some errors.
I also tried generating new certificates and the problem persists.

### How to reproduce

````
        private const string _myWebServerCrt =
@"-----BEGIN CERTIFICATE-----
MIIDNzCCAh+gAwIBAgIBAjANBgkqhkiG9w0BAQsFADA7MQswCQYDVQQGEwJOTDER
MA8GA1UECgwIUG9sYXJTU0wxGTAXBgNVBAMMEFBvbGFyU1NMIFRlc3QgQ0EwHhcN
MTkwMjEwMTQ0NDA2WhcNMjkwMjEwMTQ0NDA2WjA0MQswCQYDVQQGEwJOTDERMA8G
A1UECgwIUG9sYXJTU0wxEjAQBgNVBAMMCWxvY2FsaG9zdDCCASIwDQYJKoZIhvcN
AQEBBQADggEPADCCAQoCggEBAMFNo93nzR3RBNdJcriZrA545Do8Ss86ExbQWuTN
owCIp+4ea5anUrSQ7y1yej4kmvy2NKwk9XfgJmSMnLAofaHa6ozmyRyWvP7BBFKz
NtSj+uGxdtiQwWG0ZlI2oiZTqqt0Xgd9GYLbKtgfoNkNHC1JZvdbJXNG6AuKT2kM
tQCQ4dqCEGZ9rlQri2V5kaHiYcPNQEkI7mgM8YuG0ka/0LiqEQMef1aoGh5EGA8P
hYvai0Re4hjGYi/HZo36Xdh98yeJKQHFkA4/J/EwyEoO79bex8cna8cFPXrEAjya
HT4P6DSYW8tzS1KW2BGiLICIaTla0w+w3lkvEcf36hIBMJcCAwEAAaNNMEswCQYD
VR0TBAIwADAdBgNVHQ4EFgQUpQXoZLjc32APUBJNYKhkr02LQ5MwHwYDVR0jBBgw
FoAUtFrkpbPe0lL2udWmlQ/rPrzH/f8wDQYJKoZIhvcNAQELBQADggEBAC465FJh
Pqel7zJngHIHJrqj/wVAxGAFOTF396XKATGAp+HRCqJ81Ry60CNK1jDzk8dv6M6U
HoS7RIFiM/9rXQCbJfiPD5xMTejZp5n5UYHAmxsxDaazfA5FuBhkfokKK6jD4Eq9
1C94xGKb6X4/VkaPF7cqoBBw/bHxawXc0UEPjqayiBpCYU/rJoVZgLqFVP7Px3sv
a1nOrNx8rPPI1hJ+ZOg8maiPTxHZnBVLakSSLQy/sWeWyazO1RnrbxjrbgQtYKz0
e3nwGpu1w13vfckFmUSBhHXH7AAS/HpKC4IH7G2GAk3+n8iSSN71sZzpxonQwVbo
pMZqLmbBm/7WPLc=
-----END CERTIFICATE-----";

        private const string _myWebServerPrivateKey =
@"-----BEGIN RSA PRIVATE KEY-----
MIIEpAIBAAKCAQEAwU2j3efNHdEE10lyuJmsDnjkOjxKzzoTFtBa5M2jAIin7h5r
lqdStJDvLXJ6PiSa/LY0rCT1d+AmZIycsCh9odrqjObJHJa8/sEEUrM21KP64bF2
2JDBYbRmUjaiJlOqq3ReB30Zgtsq2B+g2Q0cLUlm91slc0boC4pPaQy1AJDh2oIQ
Zn2uVCuLZXmRoeJhw81ASQjuaAzxi4bSRr/QuKoRAx5/VqgaHkQYDw+Fi9qLRF7i
GMZiL8dmjfpd2H3zJ4kpAcWQDj8n8TDISg7v1t7HxydrxwU9esQCPJodPg/oNJhb
y3NLUpbYEaIsgIhpOVrTD7DeWS8Rx/fqEgEwlwIDAQABAoIBAQCXR0S8EIHFGORZ
++AtOg6eENxD+xVs0f1IeGz57Tjo3QnXX7VBZNdj+p1ECvhCE/G7XnkgU5hLZX+G
Z0jkz/tqJOI0vRSdLBbipHnWouyBQ4e/A1yIJdlBtqXxJ1KE/ituHRbNc4j4kL8Z
/r6pvwnTI0PSx2Eqs048YdS92LT6qAv4flbNDxMn2uY7s4ycS4Q8w1JXnCeaAnYm
WYI5wxO+bvRELR2Mcz5DmVnL8jRyml6l6582bSv5oufReFIbyPZbQWlXgYnpu6He
GTc7E1zKYQGG/9+DQUl/1vQuCPqQwny0tQoX2w5tdYpdMdVm+zkLtbajzdTviJJa
TWzL6lt5AoGBAN86+SVeJDcmQJcv4Eq6UhtRr4QGMiQMz0Sod6ettYxYzMgxtw28
CIrgpozCc+UaZJLo7UxvC6an85r1b2nKPCLQFaggJ0H4Q0J/sZOhBIXaoBzWxveK
nupceKdVxGsFi8CDy86DBfiyFivfBj+47BbaQzPBj7C4rK7UlLjab2rDAoGBAN2u
AM2gchoFiu4v1HFL8D7lweEpi6ZnMJjnEu/dEgGQJFjwdpLnPbsj4c75odQ4Gz8g
sw9lao9VVzbusoRE/JGI4aTdO0pATXyG7eG1Qu+5Yc1YGXcCrliA2xM9xx+d7f+s
mPzN+WIEg5GJDYZDjAzHG5BNvi/FfM1C9dOtjv2dAoGAF0t5KmwbjWHBhcVqO4Ic
BVvN3BIlc1ue2YRXEDlxY5b0r8N4XceMgKmW18OHApZxfl8uPDauWZLXOgl4uepv
whZC3EuWrSyyICNhLY21Ah7hbIEBPF3L3ZsOwC+UErL+dXWLdB56Jgy3gZaBeW7b
vDrEnocJbqCm7IukhXHOBK8CgYEAwqdHB0hqyNSzIOGY7v9abzB6pUdA3BZiQvEs
3LjHVd4HPJ2x0N8CgrBIWOE0q8+0hSMmeE96WW/7jD3fPWwCR5zlXknxBQsfv0gP
3BC5PR0Qdypz+d+9zfMf625kyit4T/hzwhDveZUzHnk1Cf+IG7Q+TOEnLnWAWBED
ISOWmrUCgYAFEmRxgwAc/u+D6t0syCwAYh6POtscq9Y0i9GyWk89NzgC4NdwwbBH
4AgahOxIxXx2gxJnq3yfkJfIjwf0s2DyP0kY2y6Ua1OeomPeY9mrIS4tCuDQ6LrE
TB6l9VGoxJL4fyHnZb8L5gGvnB1bbD8cL6YPaDiOhcRseC9vBiEuVg==
-----END RSA PRIVATE KEY-----";

X509Certificate _myWebServerCertificate509 = new X509Certificate2(_myWebServerCrt, _myWebServerPrivateKey, "1234");

nanoFramework.WebServer.WebServer webServer = new nanoFramework.WebServer.WebServer(443, HttpProtocol.Https, types)
{
    HttpsCert = _myWebServerCertificate509,

    SslProtocols = SslProtocols.Tls12 | SslProtocols.Tls13
};

webServer.Start()

```

```
++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (5) ++++
++++ Message: 
++++ System.Net.Security.SslNative::SecureServerInit [IP: 0000] ++++
++++ System.Net.Security.SslStream::Authenticate [IP: 0024] ++++
++++ System.Net.Security.SslStream::AuthenticateAsServer [IP: 0016] ++++
++++ System.Net.HttpListener::AcceptThreadFunc [IP: 009a] ++++
```

![Image](https://github.com/user-attachments/assets/f40bab1b-dea1-42a9-a140-2c8469d2940b)

### Screenshots

Make it possible to receive HTTPS requests

renerlemes, 2025-06-30:
> Just to male it sure I understand correctly:
> 
> * You have 1 certificate for the WebServer using HTTPS
> * You have 1 certificate for MQTTS
> * * The 2 certificates are different
> 
> If you are using one of the other, everything works fine, the problem is when you have both.
> Is that a good understanding of the situation?

@Ellerbach Yes, that is exactly what is happening.

Ellerbach, 2025-07-03:
@renerlemes Can you confirm that each part is individually working fine? And that it's only when you have both that it's not working?

josesimoes, 2025-07-03:
I was just reading this again and it stroke me: I'm not sure if the mbedTLS API is prepared to handle two different server certificates... Almost sure our API isn't. I mean, there a single execution path for the SSL that at some point setups the server cert. 
I don't think it was ever foreseen the need to have different server certs depending on the endpoint.

Wondering why can't you use the same server cert for both scenarios?

Ellerbach, 2025-07-03:
@renerlemes can you try to place both certs in the cert section? as far as I remember, mbeds can handle multiple certs but also, they should be able to be uploaded at the same time otherwise one will replace the other one. In sort, put both together, copy/paste both section together and check if still works individually and then if works together.

renerlemes, 2025-07-03:
I didn't know about this certificate part, but I can try using just one certificate to see if it solves the problem.

Regarding your question, I need to use secure access, because I have a web page that connects to the device's IP when it is in AP Mode and performs the necessary configurations for it to work the way I need it to.

For security reasons, browsers do not allow a connection between HTTPS (configuration page) and HTTP (my device's webserver).

Of course, I could change the part of my configuration page to run HTTP, but most hosting services no longer allow it without a security certificate.

alberk8, 2025-07-05:
> I was using Postman for testing and now I realized that it was ignoring the certificate. There really won't be any way to use the webserver with HTTPS, because as it is in a local environment, there is no way to generate a valid and trusted certificate (as far as I know)

If you own a domain, then you can generate a site or wildcard certificate with LetsEncrypt for free (need to renew every 3 months).  The certificate will be valid for online or offline use.

renerlemes, 2025-07-05:
You need to have a public domain to generate a valid SSL certificate. Even using mDNS as an alternative, it would still not be possible

Ellerbach, 2025-07-06:
@renerlemes you can add the certs one after the the other one like this:

        private const string _myCerts =
@"-----BEGIN CERTIFICATE-----
DATAFORFIRSTCERT
-----END CERTIFICATE-----
-----BEGIN CERTIFICATE-----
DATAFORSECONDCERT
-----END CERTIFICATE-----";

That should work fine

Ellerbach, 2026-09-28:
@renerlemes I will go ahead and close this one as opened for more than a year and we did not hear any feedback from you. In case there is still an issue, feel free to reopen an issue.

closed 2026-09-28 by Ellerbach