nanoframework/Home #1657: Modbus Max-Device-ID too low for devices with high IDs

muenchris, 2025-07-19:
### Library/API/IoT binding

Modbus

### Visual Studio version

VS2022.17.4

### Target name(s)

ALL

### Firmware version

N/A

### Description

There are devices out there (i.e. https://www.amazon.com/dp/B07ZYVZZKK?ref=ppx_yo2ov_dt_b_fed_asin_title - See section #3 "Read device ID") that use a high ID during calibration or readout of their real ID. 

        /// <summary>
        /// The highest accepted device id on RTU protocol.
        /// </summary>
        public const byte MaxDeviceId = 0xF7; // 247 
This number needs to be increased to 255 to be able to read these parameters off these devices

### How to reproduce

Try to access any device with a system parameter device ID larger than 247 like 
https://www.amazon.com/dp/B07ZYVZZKK?ref=ppx_yo2ov_dt_b_fed_asin_title

### Expected behaviour

Should be possible somehow. Maybe with a special call that allows to address these high numbers

muenchris, 2025-07-21:
I have tested this with other free Modbus testing tools like: https://www.modbustester.com/#download
they do accept higher IDs.
On the desktop I am using https://github.com/NModbus/NModbus and it does not do any checks on the slave address allowing even 255 as a valid address.

josesimoes, 2025-08-13:
I finally got the time to look into this.
Looking at the official [Modbus Serial Line Protocol and Implementation Guide V1.02](https://modbus.org/docs/Modbus_over_serial_line_V1_02.pdf), sections 2.1 and 2.2. Our implementation is correct as it validates the address being in the range [1... 247]. The spec clearly mentions that addresses 248 to 255  are reserved.

Looks like the device you're using is not following the spec...

In principle, our implementation should follow the official spec. I don't see a way of fixing this without hijacking that constant or add a new property that allows the user to shoot itself in the foot (sort of speaking) and disable the checks... 
Thoughts?

josesimoes, 2025-08-13:
### Implementation suggestion
A `bool` property added to `ModbusClient` class. Only setter. With a descriptive name, like `AllowAnyDeviceId`. IntelliSense comment describing the intent, with a remark clearly stating "Use at your own risk" or similar.
This property would be checked at the `ValidParameters` methods and skips the validation of `deviceId` parameter.

muenchris, 2025-08-13:
I can do it, but it might take some time. I am currently in Germany due to a family emergency. Once I have it resolved, I will work on this.

KiwiBryn, 2026-01-06:
@josesimoes if this is still up for grabs I will add a sensor with modbus address > 247 to my next order so I can test with real HW

josesimoes, 2026-01-06:
Hey @KiwiBryn it is. Awesome if you can tackle it!